EZ-Robotics/EZ-Template #16: "Tug-of-war" detection

Unionjackjz1, 2021-10-30:
During autonomous in Tipping Point, teams rush towards center goal.  If two opposing teams latch on, and `wait_drive()` or `wait_until()` exits due to unexpected behavior, a variable should be accessible to the user telling them this happened. 

This is an example use, where `steal` is a simple autonomous mode that drives forward, grabs the neutral goal and drives backwards.  On the drive backwards, if `wait_until()` exits in a way that triggers `tug_of_war`, the rest of the autonomous mode can be ignored and the robot will attempt to escape this tugging battle. 

```C++
void tug (int attempts) {
  for (int i=0; i<attempts-1; i++) {
    // Attempt to drive backwards
    set_drive_pid(drive, -24, 127);
    wait_drive();

    // If failsafed...
    if (tug_of_war) {
      reset_drive_sensor();
      set_drive_pid(drive, -2, 20);
      pros::delay(1000);
    }
  }
}

void steal() {

  // Drive towards center goal
  claw_up();
  set_drive_pid(drive, 45, 127);
  lift_down(true);
  // Slow down when close to center
  wait_until(40);
  set_max_speed(40);
  claw_down();
  wait_drive();

  // Drive back with goal
  set_drive_pid(drive, -36, 127, true);
  // Wait until far enough away from center,
  wait_until(-14);

  if (tug_of_war) {
    tug(3);
    return;
  }

  claw_up();
  wait_drive();

}
```

Currently the exit conditions fail-safe when they read a velocity of 0 for x amount of time, but this should be changed to if the velocity isn't close to what expected velocity would be for x amount of time.  This solves the issue of wheels skidding but the robot not moving, as if you were driving into a wall at full power.

fishsticks89, 2021-10-30:
this is quite possible; should we make wait until return a bool value or should it be drive.tugOfWarState or something

Unionjackjz1, 2021-10-30:
I like `drive.tugOfWarState` more.  I also don't know if it should be called tug of war to avoid confusion in later seasons.  

Do you think this should this return 3 values; normal exiting, weird velocity exit, and "in transit"?

fishsticks89, 2021-10-31:
drive.disturbed?
drive.offCource?
drive.interfered?
I have no idea

Unionjackjz1, 2021-11-01:
I like `.interfered`.

Also, the exit conditions should be modified so if we timeout X times in a row, the autonomous exits. 

X should be a reasonable default, probably 3, but something should exist to modify this number, or disable it entirely.

fishsticks89, 2021-11-01:
I’m not sure if there is a good way of doing that second bit… I’m not sure that it would do more good than harm cause they could just restart the program if they wanted

Unionjackjz1, 2021-11-07:
2.0.0 has added a function to change exit condition values, and they are set to default values in the drive constructor.

Unionjackjz1, 2021-11-08:
Mostly, this code works as expected.  Robot drives forward and if it isn't interfered, it turns.  During the drive forward if it exits for velocity or mA, it will `tug()`.

Need to do more testing with the drive heating up, I think this changes the max amps a motor can pull, making the constants useless after the drive is a little hot.  

```C
 void tug (int attempts) {
   for (int i=0; i<attempts-1; i++) {
     // Attempt to drive backwards
     printf("i - %i", i);
     chassis.set_drive_pid(-12, 127);
     chassis.wait_drive();

     // If failsafed...
     if (chassis.interfered) {
       chassis.reset_drive_sensor();
       chassis.set_drive_pid(-2, 20);
       pros::delay(1000);
     }
     // If robot succesfully drove back, return
     else {
       return;
     }
   }
 }

void auto1() {
  chassis.set_drive_pid(24, 110, true);
  chassis.wait_drive();

  if (chassis.interfered) {
    tug(3);
    return;
  }

  chassis.set_turn_pid(90, 90);
  chassis.wait_drive();
}
```

Unionjackjz1, 2021-11-12:
This was fixed with `is_over_current()`

closed 2021-11-12 by Unionjackjz1