EZ-Robotics/EZ-Template #45: Auton Position Tracking without Gyroscope

Asik007, 2022-01-16:
I think it would be cool if there was support for odometry that only uses tracking wheels because gyroscopes are expensive for small teams. It would kinda suck because the encoders tend to build up error but even then.

Unionjackjz1, 2022-01-16:
2.2 is going to introduce odometry, and with that functionality to not need an IMU.  We'll use this issue to tag stuff that relates to it :).  This development has already started in our odom branch

Unionjackjz1, 2022-02-03:
After discussion, we decided odometry is out of scope of this project (at least for now).  Getting angle from encoders should still be a feature added.

Unionjackjz1, 2022-08-26:
I've calculated angle on drive encoders before and didn't have great success with it.  Having the same subsystem that calculates angle/position also be whats being powered gave me some nasty oscillations that could never be solved.

I think this feature should be added but only supported with external sensors to encourage the use of tracking wheels when using this.

ssejrog, 2024-06-18:
As stated previously, using drive encoders for this is pretty finicky and I believe EZ-Template needs to prioritize using drive encoders over tracking wheels.  Tracking wheels will always be better, so EZ-Template should prioritize consistency without them, then robots will only perform better once tracking wheels are added.  

For this to be implemented, it would need to:
- work well without tracking wheels
- be extremely simple to tune (maybe even an automated tuning cycle can be made, but at minimum live track width tuning)

I am open to this being added eventually.  But I do not see this solving any problems users are having right now so this should not be a priority.

ssejrog, 2024-12-19:
With the new measure offsets function, this is possible now.  Priority of this should be much lower than #139