EZ-Robotics/EZ-Template #52: `set_tank` should disable the PID

Unionjackjz1, 2022-02-07:
in auto, if a user wants to just set power to the drive, they currently have to `set_mode(ez::DISABLED)`, then `set_tank(127, 127)`.  

set_tank should disable PID for the user, and the pid tasks should use a different function for that is something like

```cpp
private_set_tank(int left, int right) {
  if (get_mode() == ez::DISABLED) return;
  set_tank(left, right);
}
```

BDubs401, 2023-11-07:
So how would one make the robot drive say 10 inches at full speed without using PID? Tried using "move_relative" but it didn't work after running the PID. As far as I know, we can only use either pid *OR* move_relative.

Unionjackjz1, 2023-11-14:
While I haven't tested it, something like the below should work.  Setting the PID mode to `ez::DISABLE` after running PID is necessary to disable the PID task and allow you to do whatever you want with the drive motors.

```cpp
void my_move_relative(double position, double velocity) {
  for (auto i : chassis.left_motors) {
    i.move_relative(position, velocity);
  }
  for (auto i : chassis.right_motors) {
    i.move_relative(position, velocity);
  }
}

void autonomous() {
  chassis.set_drive_pid(12, DRIVE_SPEED);
  chassis.wait_drive();

  chassis.set_mode(ez::DISABLE); 

  my_move_relative(1000, 200);
}
```

closed 2024-01-29 by Unionjackjz1