EZ-Robotics/EZ-Template #59: Relative Set Angle

HakopZ, 2022-05-22:
Adding a relative set angle so that you can just say turn 45 instead of the absolute position.

mkhlb, 2022-09-01:
as far as i can tell set angle sets the gyro AND the headingPID to the same thing, so the robot expressly won't turn when you set it, no?

Unionjackjz1, 2022-09-01:
new func thats basically the exact same as set_turn_pid except its called something different (maybe set_relative_turn_pid() and the target is relative so
for example
youre facing absolute 270
set_turn_pid(90) would turn you relative 180 deg backwards to face absolute 90
set_relative_turn_pid(90) would turn you relative 90 forwards to face 360
so the target in relative is just gyroPID.target + parameter

New function for turning thats structurally identical to `set_turn_pid` except its called maybe `set_relative_turn_pid()`. 

For example:
Robot is facing absolute 270
 - `set_turn_pid(90)` would turn you relative 180 deg backwards to face absolute 90
 - `set_relative_turn_pid(90)` would turn you relative 90 forwards to face 360

The target in relative is just `gyroPID.target + parameter`.

Unionjackjz1, 2023-03-30:
This should also apply for swing turns

closed 2024-01-29 by Unionjackjz1