EZ-Robotics/EZ-Template #60: Motor temperatures on screen

Unionjackjz1, 2022-08-02:
Add a page to the brain screen that has motor temperatures.  Map motor temperatures to a color gradient and display boxes on the screen where each motor is the color of the temperature, maybe blue coldest and red hottest.  

Our code should figure out what ports are motors, display either the port number in the colored box or the user can have text in the box that they define.

nightwolf3140, 2025-01-04:
this could be an interesting troubleshooting feature, I actually was considering recently making a program or something for checking motor temps in the pits. Although not entirely sure if its super nescecary.

ssejrog, 2025-01-08:
> this could be an interesting troubleshooting feature, I actually was considering recently making a program or something for checking motor temps in the pits. Although not entirely sure if its super nescecary.

From using [EZ-GUI](https://github.com/EZ-Robotics/EZ-GUI) throughout PROS 3, I can confirm how useful it is to have motor temperatures on screen.  By default the screen was black, and as the motors got warm the screen would have more and more colored boxes.  This made it very easy to visually see that the robot was warm.  

This feature should come along with #15 for 4.0.

closed 2025-10-02 by nightwolf3140