EZ-Robotics/EZ-Template #65: Reevalute all function/variable names

Unionjackjz1, 2022-09-01:
Names weren't given much thought and should be as intuitive as possible.  This should be the final thing done along with #50 before the 3.0 beta is sent out. 

All auton selector relating things will be moved to [EZ-GUI](https://github.com/EZ-Robotics/EZ-GUI) and will be included with the example project by default.

mkhlb, 2022-09-01:
function and method documentation comments should also on the docket for reevaluation

Unionjackjz1, 2022-09-01:
`Remember 4" wheels are actually 4.125!` comments should be modified to remind new teams that only the old 4" wheels are 4.125, the new ones are 4". 

Maybe instead of modifying the comment in code, we have a place on the website to tell people what size their wheels are and how to figure out their drive ratio easier.

Unionjackjz1, 2022-09-01:
The standardization of naming should lead to the (overdue) creation of contributing guidlines.

Ryan4253, 2022-09-11:
use camalCase instead of snake_case?

closed 2024-01-29 by Unionjackjz1