EZ-Robotics/EZ-Template #70: Hello, I would like to know if your automatic functions can be run manually in a non-competitive state

mkj3085003, 2023-02-27:
#### Is your feature request related to a problem? Please describe.
A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]

#### Describe the solution you'd like
A clear and concise description of what you want to happen.

#### Describe alternatives you've considered
A clear and concise description of any alternative solutions or features you've considered.

#### Additional context
Add any other context or screenshots about the feature request here.

Unionjackjz1, 2023-03-03:
Can you clarify the problem you're having?

Mactar1233, 2023-03-03:
> Can you clarify the problem you're having?

The problem is he's trying to run autons in - the driver control period, which currently you can't do

closed 2024-01-29 by Unionjackjz1