EZ-Robotics/EZ-Template #132: Switch to LemLib Units

ssejrog, 2024-06-12:
Requiring Okapi as a dependency just for units is silly.  We should switch over to using [LemLib Units](https://github.com/LemLib/units/) eventually.  This will cause breaking changes and it's probably not worth it to release 4.0 now.  This should be added to whatever breaking changes are required for a new feature in the future.

nightwolf3140, 2025-10-25:
This would be a bit more complicated of a port than originally anticipated. Shelving this for now.

ssejrog, 2026-09-07:
Hopefully, the solution to this will be [EZ-Units](https://github.com/EZ-Robotics/EZ-Units), which is Okapi units alone. Currently untested, but will look into it more before the 4.0 full release

nightwolf3140, 2026-09-09:
I totally agree thats probably the best way for us to go forward since switching to lemlib-units would require a much more complicated restructuring, especially since most of what Okapi ships with is unnecessary.

ssejrog, 2026-09-19:
Done in #392. OkapiLib is replaced by EZ-Units (vendored at `include/EZ-Units/units.hpp`) and removed from the repo. Function calls and the `_in` / `_deg` / `_ms` literals are unchanged. Included in [v4.0.0-beta.2](https://github.com/EZ-Robotics/EZ-Template/releases/tag/v4.0.0-beta.2).

closed 2026-09-19 by ssejrog