EZ-Robotics/EZ-Template #139: Multiple IMU Support

ssejrog, 2024-06-18:
### Is your feature request related to a problem? Please describe.
As EZ-Template relies on the IMU, if it ever disconnects or causes problems, you are screwed.  

### Describe the solution you'd like
Allowing multiple IMUs fully for redundancy seems like a positive thing.  It needs to be decided if these multiple IMUs will be averaged, if it'll iterate between all of them, how IMU scaling will work, how to handle disconnecting (if we continue to read an IMU after it reconnects), there is probably more too.  

### Describe alternatives you've considered
Track width could be used when IMU fails, but I don't like this solution because it requires tuning track width.  Multiple IMUs would benefit from all being scaled correctly, but I don't think it's required.  Good results with track width require tuning, which means figuring out a way to make this easy and approachable.  Ultimately I believe this would become a more complex solution.  

### Additional context
BLRS2 used double IMUs on their 24" and won worlds.  It must be good.

7865-Owlumination, 2024-06-22:
Not entirely sure if this was included with the idea, but based on the BLRS2 reference it seems like it was. But on top of the quick swap if one disconnects It would be cool to see where both are used for more accuracy and then if one fails then it swaps to just one.

ssejrog, 2024-06-22:
So you'd want some way of logging when this event happens and make it clear to a user?  Maybe it could be as simple as changing the background color on the brain screen

jemminiz, 2024-10-12:
Why not just have the list, use the first, if any disconnect log it, and if the first one disconnects go to the first non-disconnected imu? After all, IMUs have an origin at brain startup

ssejrog, 2024-10-12:
> Why not just have the list, use the first, if any disconnect log it, and if the first one disconnects go to the first non-disconnected imu? After all, IMUs have an origin at brain startup

I think this will probably be the best approach once imu scaling is figured out.  We could implement a calibration sequence that users would run once that moves the robot and stores the data to an SD card.  I think all of the scaler values would have to be relative to imu 1, so if someone tunes the scaler value later it's only affecting imu 1 and the rest are unaffected.

closed 2025-09-30 by nightwolf3140