EZ-Robotics/EZ-Template #219: ✨[FEATURE] - Documentation/Website Search Function

nightwolf3140, 2024-12-11:
## The problem  
Without a search function on the website, many features are burried within different subpages that can be difficult to locate.

## What should be added?

Having a search function will make the documentation far more intuitive to use and will aid in the discoverability of features. For me it would decrease my need to ask about particular features in the discord server to then be linked to a subpage on the website that I had no idea existed.

ssejrog, 2024-12-11:
Docusaurus supports this, we should use what they use for their docs.  https://docusaurus.io/docs/search#using-algolia-docsearch

ssejrog, 2024-12-17:
This has been added.  It searched based on a crawler and it will crawl once a week.  I can also force it to crawl.