EZ-Robotics/EZ-Template #230: 🐛[BUG] - Project Upgrade does not work

nightwolf3140, 2024-12-17:
## Expected Behavior

For it to upgrade me 3.1.0 project to 3.2.0
## Actual Behavior

```
ERROR - __main__:main - Invalid simple block '3.2.0.' - pros-cli version:3.5.4
  File "semantic_version\base.py", line 1043, in _parse_to_clause
  File "semantic_version\base.py", line 1063, in parse
ValueError: Invalid simple block '3.2.0.'
Sentry is attempting to send 1 pending events
Waiting up to 2 seconds
Press Ctrl-Break to quit
```
## Steps to Reproduce

1. Download EZ Template 3.2.0.zip, add it to project folder
2. `pros c fetch EZ-Template@3.2.0.zip`
3. `pros c apply EZ-Template@3.2.0`

## Versions

EZ-Template  3.1.0      local
kernel       4.1.0      pros-mainline
liblvgl      8.3.8      kernel-early-access-mainline
okapilib     5.0.0      kernel-early-access-mainline

## OPTIONAL: Additional Information

I thought this was just happening to me but after talking to people on the discord server this is happening to other people.

jemminiz, 2024-12-17:
try `pros c apply kernel@latest` and then upgrade. should work fine.

nightwolf3140, 2024-12-17:
> try `pros c apply kernel@latest` and then upgrade. should work fine.

Thanks that worked. Might be good for that to be noted in upgrade docs.

PR #231 (2024-12-17): Small verbiage
## Summary:

## Motivation:

### References (optional):

Closes #230 

## Test Plan:

- [ ] test item
closed 2024-12-17 by ssejrog