EZ-Robotics/EZ-Template #250: 🐛[BUG] - Simplify adding PIDs to the PID Tuner

ssejrog, 2024-12-20:
## Expected Behavior

There should be 1 line of code to add new PIDs to the PID Tuner

## Actual Behavior

instead, you have to do this:

## Steps to Reproduce

This is not the intended way to add new PIDs.  When this feature was added the simplified PID Tuner did not exist, and this was missed.
```cpp
chassis.pid_tuner_pids.push_back({"Lift PID", &liftPID.constants});
chassis.pid_tuner_full_pids.push_back({"Lift PID", &liftPID.constants});
```

## Versions

PROS Kernel Version: 4.1.1 
EZ-Template Version: 3.2.1 
Other Libraries: 

## OPTIONAL: Additional Information

Ai80-8, 2025-07-28:
Delete it all

ssejrog, 2026-09-07:
This is already fixed on \`dev\` — \`pid_tuner_add()\` was added in #253 (commit d8ef12a) and does exactly what this issue asks for:

\`\`\`cpp
chassis.pid_tuner_add({"Lift PID", &liftPID.constants});
\`\`\`

That one call adds to both the default and full PID tuners. Closing.

closed 2026-09-07 by ssejrog