EZ-Robotics/EZ-Template #290: 🐛 - More robust 0 velocity determining

ssejrog, 2025-03-19:
## Expected Behavior

All drive speeds should work the same with velocity exiting

## Actual Behavior

Slower drives, especially slower drives that heat up, will sometimes have false positives for deciding that the chassis has stopped moving.

## Steps to Reproduce

VEXU team KUDOS expierienced this and editing how 0 velocity is determined fixed the issue for them.

## Versions

PROS Kernel Version: 
EZ-Template Version: 
Other Libraries: 
EZ 3.2.2

## OPTIONAL: Additional Information

Ai80-8, 2025-07-28:
Delete it all

ssejrog, 2026-09-21:
Notes on this from Claude Code, written up so the findings aren't lost. Nothing here has been tested on a robot, and it isn't being worked on right now.

**Status:** not fixed. #389 (fixes #289) only stops the velocity exit timer from running before the robot has moved. It doesn't change the mid-motion false positive described here.

**Two likely causes, from reading the code (unverified):**

1. **Fixed main-sensor threshold.** `velocity_zero_main` defaults to 0.05 per 10 ms pass, which is about 5 in/s for drives. A slow drive, or one whose motors are heat-limited, can stay under that for the full `velocity_exit_time` (500 ms by default) while still moving.
2. **The IMU exit is independent and can't tell cruising from stalled.** It fires when acceleration magnitude stays near 0, and steady cruising also reads about 0. A slow move with a long constant-speed stretch could trip it. Both paths print `VELOCITY_EXIT`, so the log doesn't show which one fired.

**Workaround with no library change:**

- Pass `use_imu = false` to `pid_drive_exit_condition_set` to test whether the IMU path is the cause.
- Or lower the main threshold with `chassis.leftPID.velocity_sensor_main_exit_set(x)` and the same call on `rightPID`. These setters exist on `PID` only, with no `Drive` wrappers.

**Why the defaults shouldn't change yet:** without knowing which path fires, lowering the thresholds could stop real stalls from exiting, and `pid_wait()` would hang.

**To confirm:** run a slow drive (for example `pid_speed_max_set(40)`) with the IMU exit on and off, and log which path fires.

**Possible fixes once that's known:** filter the main velocity over several passes, only count a stall when commanded output is above a floor, and/or gate the IMU path or make it opt-in. Each would be a separate PR against `dev` with a failing test first.

🤖 Generated with [Claude Code](https://claude.com/claude-code)